SlashDevin/NeoHWSerial #1: does this work with Arduino Due?

mark-hahn, 2016-05-28:
I want to write a sketch with this lib that works on all arduinos including the Due.  Is this possible?

Arm doesn't have the same uart hw, right?

SlashDevin, 2016-05-29:
Yes, the non-AVR cores (like the SAM on the Due) have different hardware interfaces, so the `Serial` et al objects are implemented with different code.  On the Due, they [appear to be implemented](https://github.com/arduino/ArduinoCore-samd/blob/df982572844f3df9bf5061ae5e322994dfdbcc0a/variants/arduino_zero/variant.h) with the [Uart class](https://github.com/arduino/ArduinoCore-samd/blob/dd6890a3782b6edf880c4d7da5e46bf77b71478b/cores/arduino/Uart.h), which just uses the [SERCOM class](https://github.com/arduino/ArduinoCore-samd/blob/dd6890a3782b6edf880c4d7da5e46bf77b71478b/cores/arduino/SERCOM.h) to do most of the work.

I do not have a Due, so I would not be able to test a SAMD version.  I could create a version, but you may need to do some debugging on your Due.  I suspect that most problems will be with the build process.  Is this something you'd like to try?

mark-hahn, 2016-05-29:
This is for a product I will be shipping.  I want to advertise that it
works on all Arduinos.  I probably wouldn't bother if I didn't want the
marketing claim.  The hardware already handles the 3V.

So yes, I will buy a Due and help support it.  I'm going to take a quick
peek at the hardware differences now.

mark-hahn, 2016-05-29:
Never mind.  I just found out that the arm processor supports DMA with
chained buffers of any size.  So there aren't even interrupts needed during
the transfer.

Thanks for replying.

closed 2016-05-29 by mark-hahn