SlashDevin/NeoHWSerial #2: NeoHWSerial conflicts with SD library

mikevanis, 2016-10-24:
Hi, I've tried to compile the following: 

`#include <NeoHWSerial.h>
#include <SPI.h>
#include <SD.h>

void setup() {
  NeoSerial2.attachInterrupt(handleRx);
  NeoSerial2.begin(9600);
}

void loop() {
}

static void handleRx(uint8_t c) {
  
}
`
The error I get is the following:
`/var/folders/4k/6h5l9jl178sb4hslwv3xp1mr0000gn/T/build2324820517714833591.tmp/core.a(HardwareSerial0.cpp.o): In function `__vector_25':
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/HardwareSerial0.cpp:48: multiple definition of `__vector_25'
/var/folders/4k/6h5l9jl178sb4hslwv3xp1mr0000gn/T/build2324820517714833591.tmp/NeoHWSerial/NeoHWSerial0.cpp.o:/Users/mikevanis/Dropbox/Arduino/libraries/NeoHWSerial/NeoHWSerial0.cpp:49: first defined here
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/../lib/gcc/avr/4.8.1/../../../../avr/bin/ld: Disabling relaxation: it will not work with multiple definitions
/var/folders/4k/6h5l9jl178sb4hslwv3xp1mr0000gn/T/build2324820517714833591.tmp/core.a(HardwareSerial0.cpp.o): In function `__vector_25':
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/HardwareSerial0.cpp:48: multiple definition of `__vector_26'
/var/folders/4k/6h5l9jl178sb4hslwv3xp1mr0000gn/T/build2324820517714833591.tmp/NeoHWSerial/NeoHWSerial0.cpp.o:/Users/mikevanis/Dropbox/Arduino/libraries/NeoHWSerial/NeoHWSerial0.cpp:49: first defined here
collect2: error: ld returned 1 exit status
Error compiling.
`

mikevanis, 2016-10-26:
@SlashDevin Have you got any clues? I'd love some of your wisdom on getting this to compile.

SlashDevin, 2016-10-26:
The conflict is from trying to use the `Serial` object somewhere in your build.  If `Serial` is used in the SD library, just replace `Serial` with `NeoSerial`, and make sure to include &lt;NeoHWSerial.h&gt; at the top of those files.  That's why I put this in the [Notes Section](https://github.com/SlashDevin/NeoHWSerial#notes):
>All parts of your sketch, including other libraries, must use `NeoSerial` instead of `Serial`, `NeoSerial1` instead of `Serial1`, `NeoSerial2` instead of `Serial2`, and `NeoSerial3` instead of `Serial3`...  If you see this error, some part of your code is still using `Serial`, `Serial1`, `Serial2` or `Serial3`. Be sure to check all your libraries.

In my version of the SD library, I just commented those lines out.  I've never used any of the SD methods that print.

closed 2016-10-27 by SlashDevin

faisalsaeed, 2020-05-24:
THANKYOU for this lib!