SlashDevin/NeoHWSerial #3: Unable to use usb serial on arduino micro with NeoSerial

amitv87, 2016-12-21:
I have this code
~~~
NeoSerial.begin(115200);
NeoSerial1.begin(115200);
~~~
The compiler says
~~~
'NeoSerial' was not declared in this scope
~~~
Is it not possible to use usb serial with this library?

SlashDevin, 2016-12-21:
Did you copy a version of NeoHWSerial files (e.g., from the 1.6.5r2 directory) to an Arduino/Libraries/NeoHWSerial directory?

Did you `#include <NeoHWSerial>` at the top of your sketch?

amitv87, 2016-12-21:
Yeah, I did that.
I do not get error for **NeoSerial1** but for **NeoSerial**.
Also found that **Serial** from the arduino framework is of type [Serial_](https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/cores/arduino/USBAPI.h#L152) and not **HardwareSerial**

SlashDevin, 2016-12-22:
Oh, it's for a Leonardo or Micro?  Issue #1 for the Due is the same kind of problem... it doesn't use HardwareSerial for the different hardware, either.  The Due version would require quite a few changes, but maybe the 32U4 platform wouldn't be as bad.  I'll take a look...

amitv87, 2016-12-22:
Cool