SlashDevin/NeoHWSerial #4: IDE version 1.8.2

jasin, 2017-03-30:
Will there be any updates for 1.8.2 or does this still work?

SlashDevin, 2017-04-01:
I haven't tried it yet.  Are you getting some error?

SlashDevin, 2017-04-02:
It's working for me...

closed 2017-04-02 by SlashDevin

jasin, 2017-04-03:
Thank you for taking the time to check this. I have tried and tried without success to transfer data at rates higher than 9600. 
I have finally come to the conclusion that I need new hardware before I can make a definitive answer. Sorry for the trouble

SlashDevin, 2017-04-03:
I had only tried 9600, so I just tried 115200 in 1.8.2.  It seems to send and receive ok.  Most of the `HardwareSerial` differences between 1.6.5 and 1.8.2 have been modest speed improvements and "idiot-proofing".

You might try the Arduino IDE version for the newest NeoHWSerial: 1.6.5r2.  Arduino IDE 1.6.5r5 is available [here](https://www.arduino.cc/en/main/OldSoftwareReleases).  If you're on Windows, just get the "zip for non-admin install", not the installer package.  Unzip it and double-click the Arduino.exe.  1.6.5r5 should open.  Then File->Open your sketch, and a new window will appear.

If it doesn't work with that version, then it is probably a problem with your hardware.  At higher baud rates, you have to be more careful about the wires connecting the two devices.  It could be the connection, not the Arduino itself.  If the two devices are MCUs, differences in clock speed (crystal frequency) could cause problems, too.

Good luck!