SlashDevin/NeoHWSerial #8: NeoHWSerial on non AVR core

Fred998, 2018-11-13:
hello SlashDevin
On my Arduino DUE NeoHWSerial does not compile, I would like to use it in your excellent NeoGPS for a GPS lap timer and data logging on SD.
Do you plan to wear NeoHWSerial on non AVR core like SAM or SAMD for arduino DUE or MKR?
How can we help you? or what are the steps to get there?

ninja2B, 2018-11-20:
Fred998 you can do that using NeoGPS but without NeoHWserial. Have you seen this [worked example for Due ](http://forum.arduino.cc/index.php?topic=419861.0) in Arduino forum? If you're interested I can post a later version there.

My understadning is SlashDevin is not planning a version of NeoHWserial for Due, as the architecture is so different.

Fred998, 2018-12-02:
Ok thanks ninja2B! I look at this.
It's a shame not to provide a version for SAM or SAMD core. it's the last board and much more powerful than the UNO, the interruption on the serial link is an important thing I find.
I do not despair to see a version for MKR one day!