SlashDevin/NeoHWSerial #13: NeoHWSerial.cpp:31:25: fatal error: util/atomic.h: No such file or directory

matburnham, 2021-03-14:
I'm trying to use `NeoHWSerial` with `NeoGPS` on an `esp8266`. I'm a bit confused which version I've got but it was installed with:
```
C:\Users\Mat\Documents\Arduino>arduino-cli lib install neohwserial
NeoHWSerial depends on NeoHWSerial@1.6.6
Downloading NeoHWSerial@1.6.6...
NeoHWSerial@1.6.6 downloaded
Installing NeoHWSerial@1.6.6...
Installed NeoHWSerial@1.6.6
```

I'm confused because this repository seems to have 1.0.5 and 1.6.5r2 directories. From https://github.com/SlashDevin/NeoHWSerial/network it looks like @JoeSchDev's fork may be the one.

I'm using the NMEA_isr example with the following added to switch to hardware Serial1?:

```
#define NEOGPS_USE_SERIAL1
#define NMEAGPS_INTERRUPT_PROCESSING
```

Compiling in the Arduino IDE:

```
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Mat\AppData\Local\Arduino15\packages -hardware C:\Users\Mat\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Mat\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Mat\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32cam -ide-version=10813 -build-path C:\Users\Mat\AppData\Local\Temp\arduino_build_80100 -warnings=none -build-cache C:\Users\Mat\AppData\Local\Temp\arduino_cache_285363 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\Mat\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\Mat\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\Mat\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=C:\Users\Mat\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.esptool_py.path=C:\Users\Mat\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=C:\Users\Mat\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -verbose C:\Users\Mat\AppData\Local\Temp\arduino_modified_sketch_179027\NMEA_isr.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Mat\AppData\Local\Arduino15\packages -hardware C:\Users\Mat\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Mat\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Mat\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32cam -ide-version=10813 -build-path C:\Users\Mat\AppData\Local\Temp\arduino_build_80100 -warnings=none -build-cache C:\Users\Mat\AppData\Local\Temp\arduino_cache_285363 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\Mat\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\Mat\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\Mat\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=C:\Users\Mat\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.esptool_py.path=C:\Users\Mat\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=C:\Users\Mat\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -verbose C:\Users\Mat\AppData\Local\Temp\arduino_modified_sketch_179027\NMEA_isr.ino
Using board 'esp32cam' from platform in folder: C:\Users\Mat\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4
Using core 'esp32' from platform in folder: C:\Users\Mat\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4
Warning: Board breadboard:avr:atmega328bb doesn't define a 'build.board' preference. Auto-set to: AVR_ATMEGA328BB
Detecting libraries used...
"C:\\Users\\Mat\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-IC:\\Users\\Mat\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/config" "-IC:\\Users\\Mat\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/app_trace" "-IC:\\Users\\Mat\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/app_update" "-IC:\\Users\\Mat\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/asio" "-IC:\\Users\\Mat\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/bootloader_support" "-IC:\\Users\\Mat\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/bt" "-IC:\\Users\\Mat\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/coap" "-IC:\\Users\\Mat\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/console" "-IC:\\Users\\Mat\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/driver" "-IC:\\Users\\Mat\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp-tls" "-IC:\\Users\\Mat\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp32" "-IC:\\Users\\Mat\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_adc_cal" "-IC:\\Users\\Mat\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_event" "-IC:\\Users\\Mat\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_http_client" "-IC:\\Users\\Mat\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_http_server" "-IC:\\Users\\Mat\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_https_ota" "-IC:\\Users\\Mat\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_ringbuf" "-IC:\\Users\\Mat\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/ethernet" "-IC:\\Users\\Mat\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/expat" "-IC:\\Users\\Mat\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/fatfs" "-IC:\\Users\\Mat\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/freemodbus" "-IC:\\Users\\Mat\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/freertos" "-IC:\\Users\\Mat\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/heap" "-IC:\\Users\\Mat\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/idf_test" "-IC:\\Users\\Mat\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/jsmn" "-IC:\\Users\\Mat\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/json" "-IC:\\Users\\Mat\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/libsodium" "-IC:\\Users\\Mat\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/log" "-IC:\\Users\\Mat\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/lwip" "-IC:\\Users\\Mat\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/mbedtls" "-IC:\\Users\\Mat\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/mdns" "-IC:\\Users\\Mat\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/micro-ecc" "-IC:\\Users\\Mat\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/mqtt" "-IC:\\Users\\Mat\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/newlib" "-IC:\\Users\\Mat\\AppData\\Local\\Arduino15\\packages\\esp32\\ [... cut ...]
Using library NeoHWSerial at version 1.6.6 in folder: C:\Users\Mat\Documents\Arduino\libraries\NeoHWSerial 
exit status 1
Error compiling for board AI Thinker ESP32-CAM.
```

It looks like the fault is casued by the addition of #include <util/atomic.h> on https://github.com/JoeSchDev/NeoHWSerial/blame/master/src/NeoHWSerial.cpp#L31 as part of 49d1112fa1e42b78114c7baa3b27c52c46bceb49. This allows the use of ATOMIC_BLOCK(ATOMIC_RESTORESTATE) but appears to be specific to AVR, and not available on ESP8266. https://github.com/wizard97/SimplyAtomic looks like a more portable way to do this.

matburnham, 2021-03-14:
It looks like the fault is casued by the addition of `#include <util/atomic.h>` on [NeoHWSerial.cpp#L31](https://github.com/JoeSchDev/NeoHWSerial/blame/master/src/NeoHWSerial.cpp#L31) as part of 49d1112. This allows the use of `ATOMIC_BLOCK(ATOMIC_RESTORESTATE)` but appears to be specific to AVR, and not available on ESP8266. https://github.com/wizard97/SimplyAtomic looks like a more portable way to do this.

Now I've dug into it, https://www.arduino.cc/reference/en/libraries/neohwserial/ suggests `This library is compatible with the avr architecture`. I guess that's my problem.

matburnham, 2021-03-14:
Hah, typical, once I remove `#include <util/atomic.h>` I lose the error and gain a rather useful warning.

`WARNING: library NeoHWSerial_dev claims to run on avr architecture(s) and may be incompatible with your current board which runs on esp32 architecture(s).`

JoeSchDev, 2021-03-16:
Hi,, so what are the next steps?

matburnham, 2021-03-16:
For now, I've avoided using NeoHWSerial and am living without interrupts. I'll revisit if I decide I need to use the extra features.

R1DEN, 2023-07-05:
@matburnham you can check https://github.com/SlashDevin/NeoGPS/issues/110#issuecomment-1622238073